yrzr/opnsense-tools #1: py39-duckdb missing from images

maurice-w, 2023-07-24:
`py39-duckdb` and some dependencies are missing from `OPNsense-23.1.11-vm-aarch64.vmdk.xz` and `OPNsense-23.7.r1-vm-aarch64.vmdk.xz`. This package is required for Unbound DNS reporting (Reporting: Settings: Unbound DNS reporting). It is available from your repository though and can be installed with `pkg install py39-duckdb`, which makes DNS reporting work.

I was just wondering if you have an idea why it's not included in the images.

Cheers & thanks for all your work!
Maurice

yrzr, 2023-07-26:
Hi Maurice. You have a good insight. I myself can hardly find a problem like this.

The problem is caused because the core only depends on `py39-duckdb` on AMD64.

https://github.com/opnsense/core/blob/77f9b43234f169cf809ec606aa06ead0de4043b2/Makefile#L131

I think it makes no sense to keep the package on amd64 only since the packages could be built on aarch64. And I have created this [pull request](https://github.com/opnsense/core/pull/6686).

yrzr, 2023-07-27:
merged

closed 2023-07-27 by yrzr

maurice-w, 2023-07-27:
Great, thanks! Looking forward to 23.7 images with duckdb.

maurice-w, 2023-08-01:
I did build an OPNsense 23.7-aarch64 VM image from source and can confirm that Unbound DNS reporting now works out of the box. 👍