yrzr/opnsense-tools #2: SSL Errors on OPNsense update

ChrisS1985ET, 2023-08-09:
Hello YRZR,

I am running the aarch64 opnsense on my espressobinv7 and all was working fine before only today I noticed i am getting the below SSL error from your repo, will this resolve itself over time? 

***GOT REQUEST TO AUDIT CONNECTIVITY***
Currently running OPNsense 23.1.11 at Thu Jul  6 01:09:44 AWST 2023
Checking connectivity for host: ftp.yrzr.tk -> 104.21.20.92
PING 104.21.20.92 (104.21.20.92): 1500 data bytes
1508 bytes from 104.21.20.92: icmp_seq=0 ttl=59 time=56.010 ms
1508 bytes from 104.21.20.92: icmp_seq=1 ttl=59 time=55.869 ms
1508 bytes from 104.21.20.92: icmp_seq=2 ttl=59 time=57.223 ms
1508 bytes from 104.21.20.92: icmp_seq=3 ttl=59 time=55.864 ms

--- 104.21.20.92 ping statistics ---
4 packets transmitted, 4 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 55.864/56.242/57.223/0.570 ms
Checking connectivity for repository (IPv4): https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1
Updating OPNsense repository catalogue...
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
pkg: https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1/latest/meta.txz: Authentication error
repository OPNsense has no meta file, using default settings
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
pkg: https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1/latest/packagesite.pkg: Authentication error
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
Certificate verification failed for /CN=yrzr.tk
1094787072:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:
pkg: https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1/latest/packagesite.txz: Authentication error
Unable to update repository OPNsense
Error updating repositories!
Checking connectivity for host: ftp.yrzr.tk -> 2606:4700:3035::6815:145c
ping: UDP connect: No route to host
Checking connectivity for repository (IPv6): https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1
Updating OPNsense repository catalogue...
pkg: https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1/latest/meta.txz: Non-recoverable resolver failure
repository OPNsense has no meta file, using default settings
pkg: https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1/latest/packagesite.pkg: Non-recoverable resolver failure
pkg: https://ftp.yrzr.tk/opnsense/FreeBSD:13:aarch64/23.1/latest/packagesite.txz: Non-recoverable resolver failure
Unable to update repository OPNsense
Error updating repositories!
***DONE***

Thanks,
Chris

ChrisS1985ET, 2023-08-09:
Also noticed autossh isnt on my install of opnsense, do you know if this pkg is available on the repo as well? Thanks for your help.

maurice-w, 2023-08-10:
`Currently running OPNsense 23.1.11 at Thu Jul 6 01:09:44 AWST 2023`

You seem to live in the past... Kidding aside: Your system time is wrong.

`autossh` isn't required by OPNsense and isn't installed by default. It is available from the repo though.

ChrisS1985ET, 2023-08-11:
Ah thank you Maurice, NTPd service had gone out of sync, all fixed now.  But just a couple questions now... Because i am using login authentication server with local+TOTP password it seems that if NTPd gets out of sync then potentially can get locked out because TOTP auth codes will not match up... have you encountered issues like this before? Is it only because these aarch64 boards do not have a clock?  Is there a way to force sync the ntpd service from console if locked out? I have disabled the "Password protect the console menu" from Administration settings, and this seems to be a good preventative measure for not getting locked out.  Or because we are using aarch64 is it advisable that TOTP not be used because ntp may potentially have sync issues?

Sorry for so many questions, this is my first time using opnsense on aarch64.

Regards,
Chris

maurice-w, 2023-08-11:
The forum might be a better place to discuss this.
(I'm using aarch64 builds for cloud VMs only, no need for NTP there.)

closed 2023-08-15 by yrzr