yrzr/opnsense-tools #4: Boot from Sata rpi4

ChrisS1985ET, 2023-08-30:
Please help, I am trying to boot your image on rpi4, have been successful booting from sdcard 'mmc' but i would like to have my u-boot boot from may sata 'scsi' connection, my attempts have not been well.  I know the Sata disk is visible from the console. Any advice? Output below. thanks.

U-Boot 2022.04-armbian (May 16 2022 - 17:50:52 +0000)

DRAM:  1 GiB
Core:  38 devices, 20 uclasses, devicetree: fit
WDT:   Not starting watchdog@8300
Comphy chip #0:
Comphy-0: USB3_HOST0    5 Gbps
Comphy-1: PEX0          5 Gbps
Comphy-2: SATA0         6 Gbps
Target spinup took 0 ms.
AHCI 0001.0300 32 slots 1 ports 6 Gbps 0x1 impl SATA mode
flags: ncq led only pmp fbss pio slum part sxs
PCIe: Link down
MMC:   sdhci@d0000: 0, sdhci@d8000: 1
Loading Environment from SPIFlash... SF: Detected w25q32dw with page size 256 By                                                                                          tes, erase size 4 KiB, total 4 MiB
OK
Model: 
Net:   eth0: ethernet@30000
Hit any key to stop autoboot:  0
MMC: no card present
Couldn't find partition mmc 0:1
Can't set block device
MMC: no card present
Couldn't find partition mmc 0:1
Can't set block device
MMC: no card present
Scanning disk sdhci@d0000.blk...
Disk sdhci@d0000.blk not ready
Found 0 disks
No EFI system partition
ERROR: invalid device tree
=> env default -a
## Resetting to default environment
=> setenv bootcmd 'scsi scan; scsi dev 0; ext4load scsi 0:0 $fdt_addr dtb/marvell/armada-3720-espressobin.dtb; ext4load scsi 0:0 $kernel_addr EFI/BOOT/bootaa64.efi; bootefi $kernel_addr $fdt_addr'
=> env save
Saving Environment to SPIFlash... Erasing SPI flash...Writing to SPI flash...done
OK
=> boot
scanning bus for devices...
  Device 0: (0:0) Vendor: ATA Prod.: SanDisk SD6SB1M1 Rev: X231
            Type: Hard Disk
            Capacity: 122104.3 MB = 119.2 GB (250069680 x 512)

Device 0: (0:0) Vendor: ATA Prod.: SanDisk SD6SB1M1 Rev: X231
            Type: Hard Disk
            Capacity: 122104.3 MB = 119.2 GB (250069680 x 512)
... is now current device
Can't set block device
Can't set block device
ERROR: invalid device tree
=>

ChrisS1985ET, 2023-09-03:
All good, I've sorted it, booting opnsense on sata but now I am getting a mount error 19 and nothings shows up as listed under geom disk listing. Any ideas?

yrzr, 2023-11-14:
I see you are using u-boot built by armbian. I haven't tried it. But I can boot my 4GB RAM RPI4 from my SSD drive through usb3 bridge.

Both u-boot and the FreeBSD 13.2 kernel can recognize my SSD drive and find the right partition. Please see your boot fails on which step.

```txt
U-Boot 2023.01 (Jul 03 2023 - 08:43:37 +0000)

DRAM:  947 MiB (effective 3.9 GiB)
RPI 4 Model B (0xc03111)
Core:  203 devices, 16 uclasses, devicetree: board
MMC:   mmc@7e300000: 3, emmc2@7e340000: 0
Loading Environment from FAT... ** Bad device specification mmc 1 **
In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 3 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found
Hit any key to stop autoboot:  0
Card did not respond to voltage select! : -110
MMC Device 1 not found
no mmc device at slot 1
MMC Device 2 not found
no mmc device at slot 2

Device 0: Vendor: GLOWAY Y Rev: F30A Prod: CT120GS3-S7
            Type: Hard Disk
            Capacity: 114473.4 MB = 111.7 GB (234441648 x 512)
... is now current device
Scanning usb 0:1...
                                                                                                                                    ^Card did not respond to voltage select! : -110
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
Found EFI removable media binary efi/boot/bootaa64.efi
1174172 bytes read in 5 ms (224 MiB/s)
Booting /efi\boot\bootaa64.efi
console comconsole failed to initialize
Consoles: EFI console
    Reading loader env vars from /efi/freebsd/loader.env
Setting currdev to disk0p1:
FreeBSD/arm64 EFI loader, Revision 1.1

   Command line arguments: loader.efi
   Image base: 0x39aff000
   EFI version: 2.100
   EFI Firmware: Das U-Boot (rev 8227.256)
   Console: efi,comconsole (0)
   Load Path: /efi\boot\bootaa64.efi
   Load Device: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b)/UsbClass(0x0,0x0,0x9,0x0,0x3)/UsbClass(0x2109,0x715,0x0,0x0,0x0)/HD(1,0x01,0,0x800,0x19000)
Trying ESP: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b)/UsbClass(0x0,0x0,0x9,0x0,0x3)/UsbClass(0x2109,0x715,0x0,0x0,0x0)/HD(1,0x01,0,0x800,0x19000)
Setting currdev to disk0p1:
Trying: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b)/UsbClass(0x0,0x0,0x9,0x0,0x3)/UsbClass(0x2109,0x715,0x0,0x0,0x0)/HD(2,0x01,0,0x19800,0x5e6800)
Setting currdev to disk0p2:
Loading /boot/defaults/loader.conf
Loading /boot/defaults/loader.conf
Loading /boot/device.hints
Loading /boot/loader.conf
Loading /boot/loader.conf.local
c\
|
             ______  _____  _____
            /  __  |/ ___ |/ __  |
            | |  | | |__/ | |  | |___  ___ _ __  ___  ___
            | |  | |  ___/| |  | / __|/ _ \ '_ \/ __|/ _ \
[... 375 lines cut ...]
Setting up routes...done.
Starting Unbound DNS...done.
Configuring firewall.......done.
Setting up gateway monitors...done.
Syncing OpenVPN settings...done.
Starting NTP service...done.
Starting Unbound DNS...done.
Generating RRD graphs...done.
>>> Invoking start script 'newwanip'
Reconfiguring IPv4 on genet0
Reconfiguring IPv6 on genet0
>>> Invoking start script 'freebsd'
Starting powerd.
>>> Invoking start script 'syslog'
>>> Invoking start script 'carp'
>>> Invoking start script 'cron'
Starting Cron: OK
>>> Invoking start script 'openvpn'
>>> Invoking start script 'sysctl'
Service `sysctl' has been restarted.
>>> Invoking start script 'beep'
>>> Error in start script '95-beep'
Root file system: /dev/ufs/OPNsense
Tue Nov 14 04:40:33 UTC 2023
```

closed 2024-04-09 by yrzr