yrzr/opnsense-tools #5: Major upgrades on repo?

ChrisS1985ET, 2023-10-02:
Can the aarch64 repo also include major updates? Is there any other way to upgrade without starting over flashing image and restoring backup?

yrzr, 2023-11-01:
Well, this is only an unofficial ARM build for OPNsense, with limited support. It is not me to decide if aarch64 is listed as a tier 1 product. However, you can make an issue and ask on the [official repo](https://github.com/opnsense/tools).

And as I have tried, restoring backups would change the booting configs in a mess way and leads to boot failure. I haven't had any time look into the problem.

I will update this issue when I have time.

mrusli70, 2024-10-14:
I did not try to use backup method. But have you try the latest version 24.7.4. The previous older version i had problems working with my raspberry pi 4. with version 24.7.4. And now i have upgraded to the latest version 24.7.6 . Anybody here knows how to resize the partition of the opnsense with growfs???

closed 2025-09-29 by yrzr