yrzr/opnsense-tools #6: Error booting in RPI4 8GB

peanutsguy, 2023-10-20:
# Basic Information

|||
|-|-|
| Device | Raspberry PI 4B 8GB |
| OS Version | OPNsense 23.7.1 for aarch64 |
| Image | OPNsense-23.7.1-arm-aarch64-RPI.img.xz |

# Reproduction Steps

Using the [article](https://www.yrzr.tk/opnsense-23-for-aarch64)

1. Downloaded the image [OPNsense-23.7.1-arm-aarch64-RPI.img.xz](https://github.com/yrzr/opnsense-tools/releases/download/23.7.1/OPNsense-23.7.1-arm-aarch64-RPI.img.xz)
2. Extracted the image `OPNsense-23.7.1-arm-aarch64-RPI.img.xz` to `OPNsense-23.7.1-arm-aarch64-RPI.img`
3. Wrote the image to a microSD
```bash
sudo dd status=progress if=OPNsense-23.7.1-arm-aarch64-RPI.img of=/dev/sdb bs=8M conv=fsync
```
4. Mounted the first partition
5. Copied the file `config_rpi4.txt` to `config.txt`
6. Unmounted the partition
7. Inserted the microSD into the RPI4 and powered on
8. Boot failed with the following [message](https://drive.google.com/file/d/1jWJ3bZ5n67Nkk8yMZbLi62SMkczb95WQ/view?usp=share_link)
```
BootOrder not defined
EFI boot manager: cannot load image
Found EFI removable media binary efi/boot/bootaa64.efi
** Reading file would overwrite reserved memory **
Failed to load 'efi/boot/bootaa64.efi'
No UEFI binary known at 0x00080000
EFI LOAD FAILED: continuing...
```

mariusherta, 2023-10-21:
I have similar issue, same device, same steps but a different [error](https://photos.google.com/photo/AF1QipPn60qMZKmyNuZsSKApWDnllameBZ-sJ-ywJVal).

yrzr, 2023-11-14:
@peanutsguy I only have a 4GB version of RPI4, this is the boot log on it with my build version 23.7.6. I wrote the image on an SSD and connected it to rpi4 through usb3 bridge.

```txt
U-Boot 2023.01 (Jul 03 2023 - 08:43:37 +0000)

DRAM:  947 MiB (effective 3.9 GiB)
RPI 4 Model B (0xc03111)
Core:  203 devices, 16 uclasses, devicetree: board
MMC:   mmc@7e300000: 3, emmc2@7e340000: 0
Loading Environment from FAT... ** Bad device specification mmc 1 **
In:    serial
Out:   vidconsole
Err:   vidconsole
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 3 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found
Hit any key to stop autoboot:  0
Card did not respond to voltage select! : -110
MMC Device 1 not found
no mmc device at slot 1
MMC Device 2 not found
no mmc device at slot 2

Device 0: Vendor: GLOWAY Y Rev: F30A Prod: CT120GS3-S7
            Type: Hard Disk
            Capacity: 114473.4 MB = 111.7 GB (234441648 x 512)
... is now current device
Scanning usb 0:1...
                                                                                                                                    ^Card did not respond to voltage select! : -110
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
Found EFI removable media binary efi/boot/bootaa64.efi
1174172 bytes read in 5 ms (224 MiB/s)
Booting /efi\boot\bootaa64.efi
console comconsole failed to initialize
Consoles: EFI console
    Reading loader env vars from /efi/freebsd/loader.env
Setting currdev to disk0p1:
FreeBSD/arm64 EFI loader, Revision 1.1

   Command line arguments: loader.efi
   Image base: 0x39aff000
   EFI version: 2.100
   EFI Firmware: Das U-Boot (rev 8227.256)
   Console: efi,comconsole (0)
   Load Path: /efi\boot\bootaa64.efi
   Load Device: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b)/UsbClass(0x0,0x0,0x9,0x0,0x3)/UsbClass(0x2109,0x715,0x0,0x0,0x0)/HD(1,0x01,0,0x800,0x19000)
Trying ESP: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b)/UsbClass(0x0,0x0,0x9,0x0,0x3)/UsbClass(0x2109,0x715,0x0,0x0,0x0)/HD(1,0x01,0,0x800,0x19000)
Setting currdev to disk0p1:
Trying: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b)/UsbClass(0x0,0x0,0x9,0x0,0x3)/UsbClass(0x2109,0x715,0x0,0x0,0x0)/HD(2,0x01,0,0x19800,0x5e6800)
Setting currdev to disk0p2:
Loading /boot/defaults/loader.conf
Loading /boot/defaults/loader.conf
Loading /boot/device.hints
Loading /boot/loader.conf
Loading /boot/loader.conf.local
c\
|
             ______  _____  _____
            /  __  |/ ___ |/ __  |
            | |  | | |__/ | |  | |___  ___ _ __  ___  ___
            | |  | |  ___/| |  | / __|/ _ \ '_ \/ __|/ _ \
            | |__| | |    | |  | \__ \  __/ | | \__ \  __/
            |_____/|_|    |_| /__|___/\___|_| |_|___/\___|

 +-----------------------------------------+     @@@@@@@@@@@@@@@@@@@@@@@@@@@@
 |                                         |   @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
 |  1. Boot Multi user [Enter]             |   @@@@@                    @@@@@
 |  2. Boot Single user                    |       @@@@@            @@@@@
 |  3. Escape to loader prompt             |    @@@@@@@@@@@       @@@@@@@@@@@
 |  4. Reboot                              |         \\\\\         /////
 |  5. Cons: Serial                        |   ))))))))))))       (((((((((((
 |                                         |         /////         \\\\\
 |  Options:                               |    @@@@@@@@@@@       @@@@@@@@@@@
 |  6. Kernel: default/kernel (1 of 1)     |       @@@@@            @@@@@
 |  7. Boot Options                        |   @@@@@                    @@@@@
 |                                         |   @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
 |                                         |   @@@@@@@@@@@@@@@@@@@@@@@@@@@@
 +-----------------------------------------+
   Autoboot in 0 seconds. [Space] to pause     23.7  ``Restless Roadrunner'' -

Loading kernel...
```

My guess is that the EFI partition on the sd-card is corrupted. Please try to write the image again, and remember to umount it carefully after modifying the  `config.txt`.

If it still cannot boot, you may also want to try whether [FreeBSD 13.2](https://download.freebsd.org/releases/arm64/aarch64/ISO-IMAGES/13.2/) works.

yrzr, 2023-11-14:
> I have a similar issue, same device, same steps but a different [error](https://photos.app.goo.gl/P8R4ptsVArtJ5DU27).

Have you changed the boot order of RPI4 through EEPROM? You need to change it accordingly.

Please try to boot [FreeBSD 13.2](https://download.freebsd.org/releases/arm64/aarch64/ISO-IMAGES/13.2/) on the same device first.

closed 2024-04-09 by yrzr