yrzr/opnsense-tools #9: Low connection speed - 300 Mbps instead of gigabit

mariusherta, 2023-11-02:
This issue occurs on all versions of opnsense. 
Steps to reproduce the bug:
- prerequisites: Raspberry Pi 4 - 4GB, laptop with gigabit ethernet, 2 micro sd cards, cat 6 ethernet cable
- on one micro sd card flash opnsense following the tutorials from https://www.yrzr.tk/, on the other micro sd card flash openwrt (in my case version 23.05)
- insert the opnsense card into the Raspberry Pi, connect the laptop to the Pi using the ethernet cable, install iperf on both the laptop and the Pi and run a test, in my case test result indicated a connection speed of ~300Mbps
- turn off the Pi, replace the opnsense micro sd card with the openwrt one, install iperf on the openwrt and run a speed test, in my case test results indicated a connection speed of ~900Mbps

yrzr, 2023-11-14:
You should enable `powerd`, which would help RPI4 reach 621-935 Mbps.

```bash
$ sysrc powerd_enable="YES"   
$ /etc/rc.d/powerd restart
```

Moreover, overclock RPI4 could also help to speed up to 799-941 Mbps.

```bash
$ cat /boot/msdos/config.txt
...
...
...
over_voltage=6
arm_freq=2000
```

And you could even disable `pf` to speed up, but is meaningless as you always want `pf` for OPNsense.

```bash
$ pfctl -d
```

Here are my tests on RPI4.

## Fresh install, FreeBSD 13.2

```bash
root@generic:~ # iperf3 -c 10.8.5.8
Connecting to host 10.8.5.8, port 5201
[  5] local 10.8.5.16 port 59531 connected to 10.8.5.8 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   113 MBytes   946 Mbits/sec   29    248 KBytes
[  5]   1.00-2.00   sec   111 MBytes   935 Mbits/sec   29   2.85 KBytes
[  5]   2.00-3.00   sec   112 MBytes   938 Mbits/sec   36   7.13 KBytes
[  5]   3.00-4.00   sec   112 MBytes   939 Mbits/sec   34    240 KBytes
[  5]   4.00-5.00   sec   111 MBytes   932 Mbits/sec   31    197 KBytes
[  5]   5.00-6.00   sec   111 MBytes   934 Mbits/sec   28    411 KBytes
[  5]   6.00-7.00   sec   112 MBytes   935 Mbits/sec   31    137 KBytes
[  5]   7.00-8.00   sec   112 MBytes   935 Mbits/sec   31    287 KBytes
[  5]   8.00-9.00   sec   112 MBytes   941 Mbits/sec   34    372 KBytes
[  5]   9.00-10.00  sec   112 MBytes   938 Mbits/sec   33   22.8 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.09 GBytes   937 Mbits/sec  316             sender
[  5]   0.00-10.00  sec  1.09 GBytes   936 Mbits/sec                  receiver

iperf Done.
root@generic:~ # iperf3 -R -c 10.8.5.8
Connecting to host 10.8.5.8, port 5201
Reverse mode, remote host 10.8.5.8 is sending
[  5] local 10.8.5.16 port 65042 connected to 10.8.5.8 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  82.2 MBytes   690 Mbits/sec
[  5]   1.00-2.00   sec  82.9 MBytes   695 Mbits/sec
[  5]   2.00-3.00   sec  82.9 MBytes   696 Mbits/sec
[  5]   3.00-4.00   sec  83.0 MBytes   696 Mbits/sec
[  5]   4.00-5.00   sec  82.8 MBytes   694 Mbits/sec
[  5]   5.00-6.00   sec  82.8 MBytes   694 Mbits/sec
[  5]   6.00-7.00   sec  83.1 MBytes   698 Mbits/sec
[  5]   7.00-8.00   sec  83.0 MBytes   696 Mbits/sec
[  5]   8.00-9.00   sec  83.0 MBytes   696 Mbits/sec
[  5]   9.00-10.00  sec  82.5 MBytes   693 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   829 MBytes   696 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   828 MBytes   695 Mbits/sec                  receiver

iperf Done.
[... 291 lines cut ...]
[  5]   0.00-10.01  sec  1.09 GBytes   940 Mbits/sec                  receiver

iperf Done.
root@OPNsense:~ # iperf3 -R -c 10.8.5.8
Connecting to host 10.8.5.8, port 5201
Reverse mode, remote host 10.8.5.8 is sending
[  5] local 10.8.5.16 port 33266 connected to 10.8.5.8 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  95.0 MBytes   797 Mbits/sec
[  5]   1.00-2.00   sec  95.4 MBytes   799 Mbits/sec
[  5]   2.00-3.00   sec  95.2 MBytes   799 Mbits/sec
[  5]   3.00-4.00   sec  95.2 MBytes   799 Mbits/sec
[  5]   4.00-5.00   sec  95.1 MBytes   798 Mbits/sec
[  5]   5.00-6.00   sec  95.1 MBytes   798 Mbits/sec
[  5]   6.00-7.00   sec  95.1 MBytes   798 Mbits/sec
[  5]   7.00-8.00   sec  95.4 MBytes   800 Mbits/sec
[  5]   8.00-9.00   sec  95.1 MBytes   799 Mbits/sec
[  5]   9.00-10.00  sec  95.2 MBytes   798 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   953 MBytes   799 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   952 MBytes   799 Mbits/sec                  receiver

iperf Done.
```

closed 2023-11-23 by yrzr

mrusli70, 2024-09-22:
Hi 

Thanks for the tips for patching up the power enable and cpu frequency. That is totally amazingly awesome!