yrzr/opnsense-tools #11: Raspberry Pi 5 Support?

dlcasto, 2023-12-03:
I am curious if the RPI image will work properly on the Raspberry Pi 5? I would be interested in trying it out if it’s possible for it to work on that hardware. I know the hardware is pretty new but they are starting to become available and possible to obtain.

Is this something that can simply be built from the tools provided? I don’t have experience building to different hardware platforms but I’m able to follow technical instruction if I have some guidance (not sure, for instance, if the Pi requires special build parameters than with other hardware platforms).

I would highlight your repo in any written or video content that I would produce if I was able to get OPNsense running on the Pi 5.

Thanks for the work in porting to other platforms!

yrzr, 2023-12-05:
We have to wait for the FreeBSD kernel to support RPI5 first.

But hopefully, we won't wait for a long time.

closed 2023-12-05 by yrzr

dlcasto, 2023-12-05:
I was afraid of that. I appreciate the response! I look forward to trying that out because the Pi 5 has a PCIe interface which could be used for a secondary network interface. I'll keep an eye on your repo if you plan to build for the Pi 5 in the future! Thanks for your work!

drduker, 2025-01-25:
any movement here?  I see that openbsd is supporting raspberry pi 5's now.  Not sure about freebsd.  No new builds since December for freebsd.

drduker, 2025-01-29:
https://forums.freebsd.org/threads/raspberry-pi-5-status.91406/post-687421

https://github.com/worproject/rpi5-uefi

Looks like with this firmware FreeBSD works fine.

yrzr, 2025-01-30:
Well, I think you can have a quick move: use my last build RPI image to flash your SD card and extract the `rpi5-uefi` firmware to the first partition (which is already formatted as FAT32 and has stock firmware). Then, you can boot the SD card with your rpi5 and load it into OPNsense. Have a good luck and let me know if it works.

yrzr, 2025-01-30:
It works, partially, however.

I made the image (check out the RPI5-UEFI image [here](https://github.com/yrzr/opnsense-tools/releases)), and it booted successfully. However, no ethernet adapter was found. As `RP1 Ethernet` is not working under `rpi5-uefi` firmware, we will need a way to bring it up.

Lalit-Patil-07, 2025-07-06:
@yrzr have u been able to test the rpi5 with your image and can you confirm its  working well ??