yrzr/opnsense-tools #13: R6C EMMC Boot

m7mdcc, 2024-01-16:
Hi , Thanks for your work 

I just wana to ask if its possible to run opnsesne along with UEFI Boot on EMMC ? i mean flash EMMC with both images ? 

Thanks

yrzr, 2024-01-17:
Have you tested with the newest UEFI image? Any results? Error logs?

As far as I tested last time, the FreeBSD kernel failed to recognize partitions on EMMC. There is still work to do on ed2k firmware.

jmccoy555, 2024-07-13:
Hi, thanks for your work!

I have managed do put UEFI on the eMMC and OPNsense on USB and UEFI on SD and OPNsense on eMMC (not a bad option really), but can't work out how to do the partitioning to go full eMMC.

Any pointers as I can't make sense of the documents?

Thanks.

mpietruschka, 2025-07-29:
Hi to everybody, what image/version of OPNSENSE are you using for your R6S? I do expirence a kernel panic and don't know what I'm doing wrong. #36