yrzr/opnsense-tools #17: Certificate verification failed for /CN=yrzr.tk

sawjai, 2024-03-19:
Unable to access  repository with following error.
Certificate verification failed for /CN=yrzr.tk
2390794240:error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed:/usr/src/crypto/openssl/ssl/statem/statem_clnt.c:1921:

It seems the certificate is expired

sawjai, 2024-03-20:
This is not an issue, you can close now

yrzr, 2024-03-28:
Sorry for the inconvenience. The problem is caused by outdated SSL certification.

As I mentioned in [this issue](https://github.com/yrzr/opnsense-tools/issues/15#issuecomment-1970259866), the domain `yrzr.tk` is no longer under my control. Therefore, I can no longer renew my SSL certification from Letsencrypt.

I have disabled the HTTPS redirect from the server. Now you can use `http://147.8.92.207/opnsense` to replace `https://ftp.yrzr.tk/opnsense` and access my repo.

closed 2024-03-28 by yrzr