yrzr/opnsense-tools #20: Kernal Panic RPI 4B

jshg46, 2024-05-21:
Anyone else had an issue with Kernal Panic on 24.1.5? New install and didn't know if this was normal.

DFS-90, 2024-06-09:
Same problem here.

Log:
```

U-Boot 2023.10 (Feb 18 2024 - 17:59:42 +0000)

DRAM:  948 MiB (effective 7.9 GiB)
RPI 4 Model B (0xd03114)
Core:  210 devices, 16 uclasses, devicetree: board
MMC:   mmc@7e300000: 3, mmc@7e340000: 0
Loading Environment from FAT... ** Bad device specification mmc 1 **
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 2 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
** Booting bootflow 'mmc@7e340000.bootdev.part_1' with efi

No EFI system partition
No EFI system partition
Failed to persist EFI variables
Booting /efi\boot\bootaa64.efi
console comconsole failed to initialize
Consoles: EFI console
    Reading loader env vars from /efi/freebsd/loader.env
Setting currdev to disk0p1:
FreeBSD/arm64 EFI loader, Revision 1.1

   Command line arguments: loader.efi
   Image base: 0x39ddc000
   EFI version: 2.100
   EFI Firmware: Das U-Boot (rev 8227.4096)
   Console: efi,comconsole (0)
   Load Path: /efi\boot\bootaa64.efi
   Load Device: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b,0000000000000000)/VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b,6a00000003000000)/SD(0)/SD(1)/HD(1,MBR,0xc27c3410,0x800,0x19000)
Trying ESP: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b,0000000000000000)/VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b,6a00000003000000)/SD(0)/SD(1)/HD(1,MBR,0xc27c3410,0x800,0x19000)
Setting currdev to disk0p1:
Trying: /VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b,0000000000000000)/VenHw(e61d73b9-a384-4acc-aeab-82e828f3628b,6a00000003000000)/SD(0)/SD(1)/HD(2,MBR,0xc27c3410,0x19800,0x5e6800)
Setting currdev to disk0p2:
Loading /boot/defaults/loader.conf
Loading /boot/defaults/loader.conf
Loading /boot/device.hints
Loading /boot/loader.conf
Loading /boot/loader.conf.local
cLoading kernel...
/boot/kernel/kernel text=0x2a8 text=0x9429e0 text=0x238bbc data=0x1b8538 data=0x0+0x2b2000 0x8+0x13ba68+0x8+0x161c23\
Loading configured modules...
umass...ukbd...usb...carp.../boot/kernel/carp.ko text=0x3500 text=0x6f80 data=0xcd0+0x48 0x8+0x19f8+0x8+0x12e9
pfsync.../boot/kernel/pfsync.ko text=0x2f63 text=0x7fcc data=0xb38+0x8 0x8+0x1848+0x8+0x1198
loading required module 'pf'
/boot/kernel/pf.ko text=0xf7e4 text=0x405dc data=0x5d20+0x43c 0x8+0x62a0+0x8+0x5171
/boot/entropy.../boot/entropy size=0x1000
pflog.../boot/kernel/pflog.ko text=0xf78 text=0x8a8 data=0x430 0x8+0xa08+0x8+0x718
ugen...can't find 'ugen'
failed!
if_vlan...if_tap...if_enc.../boot/kernel/if_enc.ko text=0x15ca text=0x934 data=0x750 0x8+0xca8+0x8+0xb71
if_bridge.../boot/kernel/if_bridge.ko text=0x3884 text=0x6e2c data=0xde8+0x8 0x8+0x1c80+0x8+0x16ac
loading required module 'bridgestp'
/boot/kernel/bridgestp.ko text=0x12e5 text=0x4cd8 data=0x2c8+0x28 0x8+0xb28+0x8+0x6af
if_lagg.../boot/kernel/if_lagg.ko text=0x3d0f text=0x9890 data=0xd10+0x8 0x8+0x1c50+0x8+0x159d
loading required module 'if_infiniband'
/boot/kernel/if_infiniband.ko text=0x1102 text=0x112c data=0x300+0x8 0x8+0x8a0+0x8+0x590
[... 319 lines cut ...]

bus_generic_new_pass() at root_bus_configure+0x40
         pc = 0xffff0000005399ec  lr = 0xffff00000053bb28
         sp = 0xffff000000fe79e0  fp = 0xffff000000fe7a10

root_bus_configure() at mi_startup+0x11c
         pc = 0xffff00000053bb28  lr = 0xffff000000469ad4
         sp = 0xffff000000fe7a20  fp = 0xffff000000fe7a50

mi_startup() at virtdone+0x78
         pc = 0xffff000000469ad4  lr = 0xffff0000000008b4
         sp = 0xffff000000fe7a60  fp = 0x0000000000000000

KDB: enter: panic
[ thread pid 0 tid 100000 ]
Stopped at      kdb_enter+0x48: undefined       f906011f
db>
db> clk_fixed4: Cannot FDT parameters.
No such command; use "help" to list available commands
db>
db> clk_fixed4: Cannot FDT parameters.
No such command; use "help" to list available commands
db> panic: vm_fault failed: ffff0000008e8684

```

DFS-90, 2024-06-09:
As mentioned [here](https://github.com/yrzr/opnsense-tools/issues/6), replacing all files in MSDOSBOOT partition with the ones from the FreeBSD image solves the problem.

I found out that replacing them with the ones from FreeBSD 14.0 leads to kernel panic.
Replacing them with those from FreeBSD 13.2 works flawlessly.

Please find the mentioned files attached:
[FreeBSD-13.2_0.fat.zip](https://github.com/user-attachments/files/15753385/FreeBSD-13.2_0.fat.zip)

Source:
[FreeBSD 13.2 RPI image](https://download.freebsd.org/releases/arm64/aarch64/ISO-IMAGES/13.2/FreeBSD-13.2-RELEASE-arm64-aarch64-RPI.img.xz)

Best regards,
David

gdman07, 2024-06-10:
I've tried replacing all the files, but this causes the pi4 to boot loop.  Could you provide instructions on which files you moved?  I get an error 110 and the pi just reboots

DFS-90, 2024-06-10:
I replaced all files on MSDOSBOOT partition with those from the "0.fat" file:
[image]

Check your config.txt - mine looks like this:
```
[all]
arm_64bit=1
dtparam=audio=on,i2c_arm=on,spi=on
dtoverlay=mmc
dtoverlay=disable-bt
device_tree_address=0x4000
kernel=u-boot.bin
enable_uart=1
uart_2ndstage=1

[pi4]
hdmi_safe=1
armstub=armstub8-gic.bin
```

459below, 2024-06-26:
Thanks @DFS-90 

I was affected by it on the RPI 4, too. Interestingly the exact same installation worked on the RPI 3 after of course changing the config.txt. I guess this is an RPI4, FreeBSD 14 upstream, regression and UBOOT thing.

Will test now and try to find the time to report back.

459below, 2024-09-06:
I just ran my compile of 24.7.3, which of course is now on FreeBSD 14.1. Both the build environment as well as the system.

The Kernel Panic issue did not occurr  to me on an entirely fresh build FreeBSD 14.1 build system with no modifications made, except for the part that was needed before, too.:
```
cd /usr/ports/ports-mgmt/pkg
make package
make reinstall
```

---

So I think the issue is now fixed. If somebody wants to reproduce, we could definitely close.

[image]

closed 2024-09-11 by yrzr