yrzr/opnsense-tools #21: 2.1 Repo - Updating OPNsense repository catalogue...

adamshumpisxxx, 2024-05-28:
I SSH into my R4S, I press 8 to enter the shell, I run your curl command, I modify the System > Firmware > Settings to (custom) as (other) isn't an option, I add "http://147.8.92.207/opnsense" without quotes and click Save. Status and Updates start spinning, Status eventually stops but Updates spin forever...Here is a screenshot of what I'm seeing:

[image]

[image]

What should I do? Thanks!

adamshumpisxxx, 2024-05-28:
Are you watching me or something...?

[image]

This thing sits for over two hours and then I post this and all of a sudden within 5 minutes it finishes! Thanks!

adamshumpisxxx, 2024-05-28:
Wait...maybe nevermind. I clicked Plugins and it went right back to spinning. 😞

yrzr, 2024-05-29:
Just have to be patient sometimes ;)

closed 2024-05-29 by yrzr