yrzr/opnsense-tools #23: unable to detect  or working with usb 3.0 gigabyte ethernet port using asix 88179 chipset with raspberry pi 4b

mrusli70, 2024-09-19:
Hi 

I unable to get the usb 3.0 gigabyte ethernet dongle to work or being detected with opnsense over my raspberry pi 4b. 

May i know what usb 3.0 gigabyte ethernet chipset or any brand that can work with opensense on raspberry pi 4b.

The asix88179 usb 3.0 gigabyte ethernet does work well on a Intel x64 PC with opnsense. But not being detected on raspberry pi 4b. 

I ran through check with hardware compatibles with freebsd that it suppose to work with freebsd. But why your opnsense for the raspberry pi 4b does not work at all with asix88179 chipset. 

Does the chipset RTL8153 work with raspberry pi4b??? Do you have any specific brand that i can try out to make it work with your software on my raspberry pi 4b.

yrzr, 2024-09-19:
Is this a USB port issue? Have you had any luck detecting other devices on that USB port?

mrusli70, 2024-09-21:
Hi yrzr, sorry for the late reply, I have tested out with USB 3.0 Gigabit Ethernet with Realtek RTL8153 chipset and finally double tested with ASIX88179 USB 3.0 Gigabit Ethernet, both works well on Raspberry pi 4b with Opnsense version 24.7.4 and updated to the latest version 24.7.4.1.  (via aarch64 automatic firmware upgrade repo walker.earth)  I double check with the Hardware Compatiblity from FreeBSD version 14.1.