yrzr/opnsense-tools #24: Opnsense 24.7.4 enable syncookies failed giving open ports tested with GRC Sheild up with Raspberry pi 4.

mrusli70, 2024-09-21:
Hi yrzr,

I tested both version of the Opnsense 24.7.4 and 24.7.4.1 , that by enable the syncookies as "always" it causing all ports to be open. 

And it failed the grc sheildup firewall stealth mode . 

But when I select the option as "never (by default)"  the syncookies  passed the grc sheildup firewall stealthj mode test. 

So I wonder if the Anti DDOS enable syncookies are having a software bug or it just a false alarm.

mrusli70, 2024-09-22:
Hi Yrzc, 
Ok I should have set the syncookies as never as default. It is actually that syncookies is enable by default set to i.  So there is no need to change anything to the Anti DDOS option. Bummer it is my fault. I am so sorry about that.

yrzr, 2024-09-22:
Hi mrusli70, I am not an expert at this. I think you should go ahead and move this issue to the official repo for help.

mrusli70, 2024-10-01:
Hi, yrzr, not too worry. I really appreciate you for having such an awesome site to have my raspberry pi 4b working with opnsense.

closed 2025-04-08 by yrzr