yrzr/opnsense-tools #25: Nanopi R4S low bandwidth 460 mbps

javierckr, 2024-10-06:
With the nanopi r4s the máximum throughput I've reached on opnsense or freebsd is 460 mbps, powerd enabled, the cpu usage is 20% Intr and 80% idle, on the same nanopi r4s with openwrt I'm getting 940 mbps.
 How could be debugged what's happening? why it's not using 100% cpu?

yrzr, 2024-10-07:
Well, this is an old topic that many may be confused about. And somehow we have discussed the network throughput performance on RPi4 [here](https://github.com/yrzr/opnsense-tools/issues/9).

The major reason leading to the poor performance from my perspective of view is that the [FreeBSD kernel does not support running clusters](https://wiki.freebsd.org/arm/RockChip).

> Currently the kernel doesn't support running clusters (big.LITTLE) at different speeds, doing so will result in (silent) crashes.
Running powerd will change frequency on demand and keep both clusters in sync.

Therefore big cores are treated equally with the little cores. The big cores are running at the same frequency as the little cores, and the kernel is unaware of that even under low frequency they are more efficient.

Even worse, as far as I know, there is no such thing in FreeBSD that is equivalent to `irqbalance` in Linux. IRQs for different devices and user programs may run on the same little core, which leads to a performance disaster.

javierckr, 2024-10-07:
Thank you, now I understand the problem, I found a command, `cpuset` https://man.freebsd.org/cgi/man.cgi?cpuset
I'll try to bind nics irqs to cpu0 and cpu1 and see if that improves something, I'll report back.

yrzr, 2024-10-07:
Also, try to bind cpu4 and cpu5 as they are big cores for RK3399.

javierckr, 2024-10-07:
Binding irqs to cpus 4 and 5 results on 780mbps still not 940, you could bind using this: https://forums.FreeBSD.org/threads/interrupts-affinity-on-arm-systems.89181/post-615309
because irqs on arm is also a little bit tricky.

closed 2024-10-08 by yrzr