yrzr/opnsense-tools #29: Zenarmor / os-sunnyvalley

bmachek, 2025-01-08:
Are you planning to release the os-sunnyvalley package in order to install the Zenarmor NGFW subsystem?
According to Zenarmor there is a aarch64 FreeBSD release of the software. :-)

Many thanks in advance.

yrzr, 2025-01-14:
Okay, let me see what I can do about it.

yrzr, 2025-01-15:
Hi @bmachek , I have added os-sunnyvalley to 24.7.11 on [my repo](http://168.138.176.159/opnsense/) and installed the package successfully.

[image]

But I am not familiar with it, please try and see whether it works and give me some feedbacks

yrzr, 2025-01-15:
Well, it seems zenarmor is not providing aarch64 packages for FreeBSD 14. Feel free to reach me again when the situation changes.

closed 2025-01-15 by yrzr