yrzr/opnsense-tools #33: LAN seems to be uninitialized in image 25.1.3 on NanoPi R4S

attilabalazsy, 2025-04-07:
My new NanoPI R4S just arrived and wanted to deploy the latest image (OPNsense-25.1.3-arm-aarch64-R4S.img.xz). The boot seems to be OK, but the LAN adapter is unresponsive. The WAN seems to be doing something, as when I connect my laptop to it, it blinks. When I do the same on LAN part, the adapter on R4S is inactive, no blinking, nothing.

Do you have any idea or suggestion, how to proceed to find the reason and correct it? Unfortunately I have no other debug information ...

yrzr, 2025-09-27:
Hi @attilabalazsy, I think mpietruschka has gone through a similar issue as yours #36 , caused by Realtek kernel drivers.

You may try to solve the problem by installing net/realtek-re-kmod198 from the FreeBSD port instead of net/realtek-re-kmod from the OPNsense port. Also see https://github.com/yrzr/opnsense-tools/issues/36#issuecomment-3146071004 as a reference.

attilabalazsy, 2025-09-29:
Hi @yrzr,

Thank you very much for your comment. The resolution looks quite easy, but R4S does not have a display port, so I cannot connect a monitor to it. And as the LAN is not working, WAN is blocking SSH, so I cannot remote to it. Do you plan to include the new driver in your build?

Thanks!

yrzr, 2025-09-29:
Yes, I have included `realtek-re-kmod198` in the newly built 25.7 versions, but it is not installed by default. Do you have a TTY to USB serial cable? You might try to connect to the OPNsense console through that.

attilabalazsy, 2025-09-29:
Yes, I have. When I connect, I see that the boot is successfully done, but then the console stops at point below I cannot cannot type anything. Do you have any suggestion?

[image]

yrzr, 2025-09-30:
Please check your console settings and also the serial speed.

[image]

attilabalazsy, 2025-10-03:
I'm using these settings:

[image]

When I switch the speed to 115200 I do not see a readable text. During the boot I can hit a key and stop the booting, so I think the communication (RX/TX) works.

attilabalazsy, 2025-10-06:
Just FYI: I managed to fix the networking by the following method:
- During the first boot, I've jumped into the process, which auto-assigns the network interfaces. I did it manually and assigned LAN to  dwc0 as this interface was working, but was auto assigned to WAN. 
- Then I was able to SSH into the device, where the menu was working 
- I've applied the driver change as described by @mpietruschka here https://github.com/yrzr/opnsense-tools/issues/36#issuecomment-3146071004
- Then I was able to change the network assignment back to re0

Now, the networking is working well, but the UART console is still not showing the menu, but this is not a prio now. Thank you for your help!

closed 2025-10-06 by attilabalazsy

yrzr, 2025-10-06:
For your UART console, could you show your console settings as in my previous screenshot?