yrzr/opnsense-tools #34: 25.1.3 RPI 4B getting stuck booting

cyberpunked1985, 2025-05-01:
25.1.3 RPI 4B getting stuck booting 
using Image [OPNsense-25.1.3-arm-aarch64-RPI.img.xz](https://github.com/yrzr/opnsense-tools/releases/download/25.1.3-yrzr/OPNsense-25.1.3-arm-aarch64-RPI.img.xz)
gets stuck at "Dual Console: Serial Primary, Video Secondary" not sure why, anyone else getting this?

cyberpunked1985, 2025-05-03:
Sorted it out, At the bootloader select 3 to go to the prompt, type 'set boot_serial=NO', type menu to go back, select 1 - worked for me. But Booting takes ages, like 20-30 minutes.  Anyone else getting super slow booting? Any way to speed it up?

Lalit-Patil-07, 2025-05-19:
can u explain this a bit i am stuck on the same issue. 
I did setup the 'set boot_serial=NO' but cant figure out to exit that section.

closed 2025-05-19 by yrzr