yrzr/opnsense-tools #36: nanopi r6s panicing as soon a network cable is connected

mpietruschka, 2025-07-07:
For some reason my opnsense is having a kernal panic as soon I connect a network cable.  For once it worked, but I cant explain why. 

I'm using [edk2-rk3588](https://github.com/edk2-porting/edk2-rk3588) revision 6eeff50 like in your blog.
OPNsense 25.1.6 is installed on an SDCard. Together with my  keyboard its connected to an usb 3.0 hub.

Sadly I can't capture the output on my monitor. It's shown to fast. I would have to connect an UART cable.

Do you have any initial Idea? Am I maybe doing something absolutely different to your setup?

mpietruschka, 2025-07-13:
I wasn't able to get the EFI console to work, so I recorded the panic. I'd appreciate any help.

https://github.com/user-attachments/assets/1d044fcb-20f3-401b-afe0-db589e7ca7b3

[image]

mpietruschka, 2025-07-13:
I found the source of the problem. The realtek vendor drive seems to be broken since version 199.

  https://forums.freebsd.org/threads/problems-with-net-realtek-re-kmod-and-freebsd-14-0.91914/
  https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=275882

Version 198 seemed to be working. So it does in your first image you created for the nanopi r6s: "OPNsense-23.7-arm-aarch64-R6S.img" .

With "OPNsense-23.7.12-arm-aarch64-R6S.img" you already started using the version 199, which ends in the above kernal panic.

The solutions from the bugtracking didn't help in my case:

  ifconfig_re0="-rxcsum6 -txcsum6 -rxcsum -txcsum DHCP"
  ifconfig_re1="-rxcsum6 -txcsum6 -rxcsum -txcsum DHCP"

Are you able to build/install the version 198 into the latest opnsense versions?

mpietruschka, 2025-07-29:
To make sure its not the null MAC Address I programmed a MAC-Address to both Realtek network chips.

https://github.com/redchenjs/rtnicpg

But I still get the kernal panic.

Any hints or ideas?

mpietruschka, 2025-07-30:
@yrzr Looks like an kernel and driver problem to me, too. 

I was able to get a working system with "OPNsense-23.7-arm-aarch64-R6S.img" . It contained the realtek driver version 198. 

> Version 198 seems to be working. So it does in your first image you created for the nanopi r6s: "OPNsense-23.7-arm-aarch64-R6S.img" .

@yrzr I'm settled in linux and new to FreeBSD. What steps would I need to do, to integrate the realtek driver version 198 into the current version of OPNsense (25.1.9)? At the moment its using version 1.100. Which is very often mentioned causing problems

mpietruschka, 2025-08-02:
There is more good news. I have managed to get the edk2-rk3588 EFI firmware running in the current version 1.1. 

[image]

Before the OS can be started, the Config Table Mode must be set to ACPI [[1]](https://forums.freebsd.org/threads/how-to-run-freebsd-on-new-boards-built-on-rockchip-35xx.88430/page-3#post-619166).

With the Realtek driver version 198, the current version of OPNSense (OPNsense-25.1.9-arm-aarch64-R6S.img) then starts without any problems. It even performs slightly better.  The unchanged OPNsense-25.1.9-arm-aarch64-R6S.img will end up with a kernel panic 😔

[image]

To speed up the boot process, I also disabled the network stack. I didn't test it further, but disabling the EFI memory attribute protocol didn't cause any problems. That's why I disabled it too.

[image]
[image]

To ensure that the system always boots directly from the USB stick or hard drive, I have disabled the other boot options, too. Apart from that, they do not yet work for FreeBSD. [[2]](https://forums.freebsd.org/threads/how-to-run-freebsd-on-new-boards-built-on-rockchip-35xx.88430/page-3#post-619166)

[image]

Links

- Forum: EDK2 Tianocore UEFI is currently available for rk3588 [[1]](https://forums.freebsd.org/threads/how-to-run-freebsd-on-new-boards-built-on-rockchip-35xx.88430/page-3#post-619166)
-  Forum: Why eMMC does not work [[2]](https://forums.freebsd.org/threads/how-to-run-freebsd-on-new-boards-built-on-rockchip-35xx.88430/page-3#post-619166)

yrzr, 2025-08-28:
@mpietruschka you can get plugins from here http://168.138.176.159/opnsense/

mpietruschka, 2025-08-28:
@yrzr Thanks, but could you maybe add them to the packages-25.1.9-aarch64.tar on the releasepage, too?

yrzr, 2025-09-24:
Hi @mpietruschka. Sorry to reply so late, I've had some busy days.

You are right about the 25.1.9 version I built. I think the 'make packages' step did not work properly during the build, so the os-* packages are not generated. And I think #37 is also related.

I am building version 25.1.12 now. And I will fix version 25.1.9 later.

yrzr, 2025-09-26:
Build version 25.1.9 has been revised and uploaded. Please take a look.

closed 2025-09-26 by yrzr

attilabalazsy, 2025-09-29:
Hi @mpietruschka ,

It seems, that I have run into the same issue with my NanoPi R4S device. I'm trying to apply the fix you have found, but unfortunately my device does not have a display port, so I have to use UART (TTY to USB). I'm not sure if, I'm doing something wrong, but for me the console (Putty) stops at this point below, and I cannot get the cursor, I cannot write. Do you have any idea? 

[image]

mpietruschka, 2025-10-02:
@attilabalazsy Hey! Usually you press enter to open the menü or login. Are you sure, that TX is connected to the right (pin, level)? Do you have a LED on your UART-Adpater showing that your keystrokes are send?

attilabalazsy, 2025-10-03:
@mpietruschka thanks for the hints. The connection should be fine as during the boot, I can type, I can stop the boot. But once it reaches the final step, I cannot type anymore. It seems like it is frozen. This is the serial config in putty, which I use:

[image]

`
U-Boot TPL 2025.04 (Aug 21 2025 - 22:34:52)

lpddr4_set_rate: change freq to 400MHz 0, 1
Channel 0: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
Channel 1: LPDDR4, 400MHz
BW=32 Col=10 Bk=8 CS0 Row=15 CS1 Row=15 CS=2 Die BW=16 Size=2048MB
256B stride
lpddr4_set_rate: change freq to 800MHz 1, 0
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2025.04 (Aug 21 2025 - 22:34:52 +0000)
Trying to boot from MMC2
## Checking hash(es) for config config-1 ... OK
## Checking hash(es) for Image atf-1 ... sha256+ OK
## Checking hash(es) for Image u-boot ... sha256+ OK
## Checking hash(es) for Image fdt-1 ... sha256+ OK
## Checking hash(es) for Image atf-2 ... sha256+ OK
## Checking hash(es) for Image atf-3 ... sha256+ OK
## Checking hash(es) for Image atf-4 ... sha256+ OK
load_simple_fit: Skip load 'atf-5': image size is 0!

U-Boot 2025.04 (Aug 21 2025 - 22:34:52 +0000)

SoC: Rockchip rk3399
Reset cause: POR
Model: FriendlyElec NanoPi R4S
DRAM:  4 GiB (effective 3.9 GiB)
PMIC:  RK808
Core:  300 devices, 32 uclasses, devicetree: separate
MMC:   mmc@fe320000: 1
Loading Environment from MMC... Reading from MMC(1)... *** Warning - bad CRC, using default environment

In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Model: FriendlyElec NanoPi R4S
Net:   eth0: ethernet@fe300000

Hit any key to stop autoboot:  0
=>
=>
=>
=>
=> kjnn
Unknown command 'kjnn' - try 'help'
=> help
?         - alias for 'help'
base      - print or set address offset
bdinfo    - print Board Info structure
....
`